pelrun/hp-omen-linux-module #1: Improve README for collaboration

alexmik95, 2020-07-05:
Hello @pelrun, I would be interested in contributing but the README don't say much.. Information about how to install and test would be helpful. On top of this I could develop an interface like HP COMMAND CENTER with GTK/TK libraries.

pelrun, 2020-07-06:
Yeah, sorry about that. In it's current form it's not usable by anyone who isn't intimately familiar with working on kernel modules, so instructions aren't going to help. I need to at the very least configure it for DKMS so it can automatically rebuild on kernel updates.

hockeymikey, 2020-12-20:
I disagree, it was pretty easy and I am not super familiar with the kernel. @alexmik95 follow the steps here: https://github.com/pelrun/hp-omen-linux-module/issues/3 skip the configure part. turn the ko into a ko.xz with xz <file>. Super easy.

closed 2021-05-07 by pelrun