pelrun/hp-omen-linux-module #2: ITS BEEN 5 MONTHS COULD YOU HURRY UP PLIZ.

Manhanted, 2020-09-11:
I just ran out of patience i wanted to switch to linux completely only to find out playibg games does not take the fans to max automatically.I looked for many ways but could not find one as convincing as this one.And plz add installation instructions.

markozajc, 2020-09-13:
You're free to do it yourself with that attitude. Remember that this is a free and opensource project, made entirely by someone's free time.
As stated in the readme (which I assume you haven't read),
> It's totally experimental right now, and could easily crash your machine. As such I'm not going to provide any instructions just yet.
If you know how kernel modules work, you'll figure it out. If you don't, you probably shouldn't try to manually replace one.

geekq, 2020-09-14:
@Manhanted At least for my Model, HP OMEN 870-155ng, the fans are sufficiently controlled by BIOS / firmware. No special kernel module necessary.
Nvidia graphics card (GeForce GTX 1070) cooling/fans can be additionally tuned via `nvidia-settings` with proprietary drivers. (installed with `sudo apt-get install nvidia-settings`, but check first if you have a Nvidia GPU)

I can confirm that when all 8 CPUs are fully loaded (when e.g. compiling) the CPU fan goes up.
When playing with 4K resolution the graphics card fan goes to the max as expected. If not much GPU power is used, the fan stays low.

I am observing this kernel module repo more for optional fun stuff like controlling ambient light etc. And installation and deinstallation instructions for just trying things out, or at least a link to a suitable kernel module instructions, would be nice! Thanks @markozajc !

markozajc, 2020-09-14:
@geekq The installation process is not too hard, but there's no automatic way to do it. You must first make the module, then replace your current kernel's hp-wmi.ko (forgot where it is, you can just find -name it) with the built one, and reboot. The zone device in /sys takes a string hex color value (0000FF) is blue for example.

pelrun, 2020-09-29:
My machine doesn't have an ambient light sensor, so it's unlikely I'll be able to do anything about that even if I can get back to working on this. Of course being shouted at is obviously the best way to motivate a person... :unamused:

closed 2021-04-20 by pelrun