pelrun/hp-omen-linux-module #3: Different color codes for HP OMEN 870-155ng ?

geekq, 2020-09-25:
I have a HP tower "HP OMEN 870-155ng".
I've installed the module as follows:

(tested with Ubuntu Mate 20.04)

```
git clone https://github.com/pelrun/hp-omen-linux-module.git
cd hp-omen-linux-module
configure
make
sudo cp /usr/lib/modules/`uname -r`/kernel/drivers/platform/x86/hp-wmi.ko \
        /usr/lib/modules/`uname -r`/kernel/drivers/platform/x86/hp-wmi.ko.original
sudo cp hp-wmi.ko /usr/lib/modules/`uname -r`/kernel/drivers/platform/x86/hp-wmi.ko
sudo reboot
```

Now I can see the devices:

```
$ ls /sys/devices/platform/hp-wmi/rgb_zones/
zone00  zone01  zone02  zone03

$ cat /sys/devices/platform/hp-wmi/rgb_zones/*
red: 0, green: 0, blue: 0
red: 0, green: 0, blue: 0
red: 0, green: 0, blue: 0
red: 0, green: 0, blue: 0
```

While by default the front panel light rotates colors changing it every couple of seconds,
writing to the zone device definitely changes the behavior.
So the new module somehow works!

```
echo '00FF00' > /sys/devices/platform/hp-wmi/rgb_zones/zone00
echo '00FF00' > /sys/devices/platform/hp-wmi/rgb_zones/zone01
echo '00FF00' > /sys/devices/platform/hp-wmi/rgb_zones/zone02
echo '00FF00' > /sys/devices/platform/hp-wmi/rgb_zones/zone03

# cat /sys/devices/platform/hp-wmi/rgb_zones/*
red: 0, green: 255, blue: 0
red: 0, green: 255, blue: 0
red: 0, green: 255, blue: 0
red: 0, green: 255, blue: 0
```

Now the color rotation has stopped and the light has turned to pure red.
But I would like to have it green!
Looks like color encoding somehow differs for my model?
Do you have any ideas how to reverse engineer the values for my device or what to try out?

pelrun, 2020-09-29:
There's a lot of Omen models and they all behave differently, so there's a whole bunch of logic in Omen Command Center just to figure out how exactly to control the device it's running on. Half the reason I haven't done a lot with this is simply because I have to replicate that logic and it's really dull! So it doesn't surprise me if things don't work right on any other model than the one I have.

geekq, 2020-09-29:
So have you got a hint for me, how to reverse engineer / find out addresses / values for my model?

How did you do for yours?

P.S. I still have a Windows installation on that machine if it is required, but no developer tools / debugger for Windows.

pelrun, 2020-09-29:
Omen Command Centre is a C# application, so it decompiles extremely cleanly with ILSpy. Reading the resulting decompiled source code provided all the details.

pelrun, 2020-09-29:
There's a half-dozen different lighting types all referred to via codenames (Drax/Typhon/Perseus/Orisa/Tracer/aaargh) so it helps to know what your machine is identified as. You can do this by looking for the product id using `sudo dmidecode` and looking for something like the following:

```
Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: HP
        Product Name: 8600
```

The 4 digit number after Product Name is what OCC uses, and it looks it up in a json file buried in HP.Omen.DeviceLib.dll.

markozajc, 2020-09-29:
> So it doesn't surprise me if things don't work right on any other model than the one I have.
@pelrun Which one do you have? I can confirm that the LED control works perfectly fine on my 15-dc1031nm (other than middle and right zones having their colors a bit off, such as having a green tint on 0xFFFFFF, but I haven't confirmed whether that's a hardware issue, something to do with a difference brightness since I'm not able to change it, or something to do with color control in the module).

pelrun, 2020-09-29:
I've got a 15-dh0146tx laptop, which has the FourZone lighting type. I haven't fiddled with the keyboard much of late (I'd rather use my mechanical while I'm stuck working from home) but I wouldn't be surprised if the colours are a bit off when controlling them directly. RGB LEDs aren't great for colour accuracy, and the OCC application might be doing colour correction itself.

pelrun, 2021-05-02:
@geekq, I've recently had some success coming at the lighting from the other direction, decompiling the ACPI bytecode that is the layer between the kernel module and the hardware itself - it turns out to be a hell of a lot easier than trying to figure things out from the application. If you want to give me a dump of your acpi tables, I can have a look at them and see how it differs from mine and @wl2776's machines.

The command is `sudo acpidump > acpidump.hex`, you can zip up and email the resulting file to me at pelrun@gmail.com.

ivan-boginski, 2021-07-24:
@pelrun  This is mine omen laptop 15 2020, I give you a link of my acpidump. 
here's the link https://drive.google.com/file/d/1lT7YJRRpXRbnXje1crTo9eFB8K5nxiu5/view?usp=sharing

pelrun, 2021-07-24:
@ivan-boginski Thanks! Is there something not working as expected on your unit,? Your dump indicates it should support the exactly the same LED interface that the module is using.

ivan-boginski, 2021-07-24:
@pelrun Thanks, I didn't try actually. I have tried, now it works perfectly, Someday I will probably make a simple gui app to control this without googling about the colors.
I was looking for how to detect CPU fan on this machine with Ryzen 4800h I wasn't able to detect this Fan sensor at least  to monitor it.

Gilgamesh0028, 2023-08-22:
the current location for Command Centre & its' modules is as follows:
```
C:\Program Files\WindowsApps\AD2F1837.OMENAudioControl_1.36.274.0_neutral_~_v10z8vjagбke6
C:\Program Files\WindowsApps\AD2F1837.OMENAudioControl_1.36.274.0_neutral_split.scale-100_v10z8vjagбke6
C:\Program Files\WindowsApps\AD2F1837.OMENAudioControl_1.36.274.0_neutral_split.scale-125_v10z8vjagбke6
C:\Program Files\WindowsApps\AD2F1837.OMENAudioControl_1.36.274.0_neutral_split.scale-150_v10z8vjagбke6
C:\Program Files\WindowsApps\AD2F1837.OMENAudioControl_1.36.274.0_x64_v10z8vjagбke6
C:\Program Files\WindowsApps\AD2F1837.OMENCommandCenter_1101.2211.4.0_x64_v10z8vjagбke6
C:\Program Files\WindowsApps\AD2F1837.OMENCommandCenter_1101.2306.4.0_x64_v10z8vjagбke6
C:\Program Files\WindowsApps\AD2F1837.OMENLightStudio_1.0.19.0_x64_v10z8vjagбke6
```

Gilgamesh0028, 2023-08-22:
@pelrun my model is an 15z-en100 2L1F0AV with FourZones lighting. 5800hx w/3070 running win10. 
What files did you use to decompile the RGB instructions?
[OpenRGBFuckery.zip](https://github.com/pelrun/hp-omen-linux-module/files/12412258/OpenRGBFuckery.zip)

Currently working with @CalcProgrammer1 on [OpenRGB-Issue790](https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/790), and need some files for OpenRGB to work with this model.

So far, I've found that the LightStudio code looks like so: [OpenRGBFuckery.zip](https://github.com/pelrun/hp-omen-linux-module/files/12412261/OpenRGBFuckery.zip)