pelrun/hp-omen-linux-module #4: Omen 15 2020 support

hockeymikey, 2020-11-29:
What's the status on the new Omen models? Some codes for them are the 15-en0029nr and the 15-en0036nr.

hockeymikey, 2020-12-20:
Confirmed working with current code.

closed 2020-12-20 by hockeymikey

piramiday, 2020-12-21:
hey @hockeymikey, are you able to provide some instructions?

hockeymikey, 2020-12-21:
@piramiday cd into the module folder and run:
`make
xz -f hp-wmi.ko
sudo cp /usr/lib/modules/`uname -r`/kernel/drivers/platform/x86/hp-wmi.ko.xz      /usr/lib/modules/`uname -r`/kernel/drivers/platform/x86/hp-wmi.ko.xz.original
sudo cp hp-wmi.ko.xz /usr/lib/modules/`uname -r`/kernel/drivers/platform/x86/hp-wmi.ko.xz
`
reboot
Changing colors

`echo 'FFFFFF' > /sys/devices/platform/hp-wmi/rgb_zones/zone0X`

piramiday, 2020-12-22:
thanks for the reply.
I have a distro based on Ubuntu 20.04, for which the `hp-wmi` file seems to be here, instead of `/usr/lib`:
```
/lib/modules/$(uname -r)/kernel/drivers/platform/x86/hp-wmi.ko
```
backing the file up and moving the new copy over is trivial.

instead I'd appreciate some more background on the compilation phase.
this is what I got from your reply. first:
```
git clone https://github.com/pelrun/hp-omen-linux-module.git
cd hp-omen-linux-module/module
```
and then
```
make
```
but I'm not sure what to do about this last `make` command.
will this put files anywhere?
will it put compiled modules under `/lib/modules/$(uname -r)/build`, which is the `KDIR` variable in the `Makefile`?
or will it just generate a new `hp-wmi.ko` module in the current folder?
I'd like to fully understand the implications before executing the command.

finally, what is the `xz` invocation for? is it needed?
thanks!

PS: to open or close multiline code blocks, put three back-quotes on a line by themselves, e.g. ```

hockeymikey, 2020-12-22:
> hp-wmi.ko module in the current folder

Just execute it, it does that. 
Compression was needed for my hp-wmi.ko since it was hp-wmi.ko.xz but I am on Arch.