pelrun/hp-omen-linux-module #5: install instructions

JavierOramas, 2021-01-14:
hi, can you give a install pathway

markozajc, 2021-01-14:
https://github.com/pelrun/hp-omen-linux-module/issues/2#issuecomment-691959312 if that helps

hockeymikey, 2021-01-14:
https://github.com/pelrun/hp-omen-linux-module/issues/3 helps more

JavierOramas, 2021-01-14:
im having this issue (im using manjaro 20.1.1):

make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: se entra en el directorio '/usr/lib/modules/5.8.11-1-MANJARO/build'
make[1]: *** No se especificó ningún objetivo y no se encontró ningún makefile.  Alto.
make[1]: se sale del directorio '/usr/lib/modules/5.8.11-1-MANJARO/build'
make: *** [Makefile:10: default] Error 2

hockeymikey, 2021-01-14:
Run it in the module folder

JavierOramas, 2021-01-14:
That's where i ran it, i found an error when creating the configure file, it couldn find configure.ca

hockeymikey, 2021-01-15:
Oh, now I see. No just run make, don't add that other crap.

JavierOramas, 2021-01-16:
it throw this error https://github.com/pelrun/hp-omen-linux-module/issues/5#issuecomment-760401876

hockeymikey, 2021-01-17:
@JavierOramas I told you, don't run with that parameter, just make.

JavierOramas, 2021-01-17:
[image]
this is what i did...

hockeymikey, 2021-01-18:
@JavierOramas You need your kernel source. I'm using the znver2 kernel from AUR so I have them from having to compile it.

pelrun, 2021-01-25:
The kernel source isn't required, just the headers.

pelrun, 2021-05-07:
Handled by DKMS now.

closed 2021-05-07 by pelrun