pelrun/hp-omen-linux-module #6: Is this project dead?

DuchGhast, 2021-03-10:


geekq, 2021-03-11:
@DuchGhast What are you trying to achieve with that question?

There is source code, there are (some) instructions, there is information on how to adjust it for a new HP model, which the author does not have. It is not much, but trolling the project creator will unlikely help!

We are on github here, so either provide a pull request with your desired improvements or ask a specific questions including detailed description what you've done, your logs etc.

markozajc, 2021-03-11:
This person has a history of opening the exact same issue on a number of projects, not sure if it's meant as a joke or something else.

pelrun, 2021-04-20:
What does it even mean anyway? Stuff doesn't happen until it happens.

closed 2021-04-20 by pelrun