pelrun/hp-omen-linux-module #9: Improve hotkey support

pelrun, 2021-05-02:
Thanks to @wl2776 I've now got ACPI dumps from two different machines and I think I can properly implement Omen key detection that will work across the board.

Depending on the key and the model, a hotkey press will generate either an 0x04 (HPWMI_BEZEL_BUTTON) or 0x1D (Omen hotkey)  WMI event, or (in my case) just raise a notification with no event data at all. In all cases, HPWMI_HOTKEY_QUERY needs to be called (as is already done for HPWMI_BEZEL_BUTTON) to get the ID of the key pressed.

The Omen key is 0x21A5, although wl2776's machine also appears to generate a 0x21A7 for some other key.

wl2776, 2021-05-02:
> wl2776's machine also appears to generate a 0x21A7 for some other key.

How did you know that?

pelrun, 2021-05-02:
Because there's both `Notify (0x1D, 0x21A5)` and `Notify (0x1D, 0x21A7)` commands in the acpi dump you sent me, obviously :)

wl2776, 2021-05-02:
I've pulled your commits, there are still issues.

If I run `sudo showkey -k` and press <kbd>Omen</kbd>  key, I see 
```
keycode 240 press
keycode 240 release
```

However, `dmesg` shows the following
```
[14285.411928] hp_wmi: Unknown key code - 0x21a5
[14285.980825] hp_wmi: Unknown key code - 0x21a5
[14286.363530] hp_wmi: Unknown key code - 0x21a5
[14439.292467] keyboard: can't emulate rawmode for keycode 240
[14439.292489] keyboard: can't emulate rawmode for keycode 240
[14439.292493] hp_wmi: Unknown key code - 0x21a5
[14439.989045] keyboard: can't emulate rawmode for keycode 240
[14439.989071] keyboard: can't emulate rawmode for keycode 240
[14439.989076] hp_wmi: Unknown key code - 0x21a5
[14440.381164] keyboard: can't emulate rawmode for keycode 240
[14440.381188] keyboard: can't emulate rawmode for keycode 240
[14440.381192] hp_wmi: Unknown key code - 0x21a5
[14440.604816] keyboard: can't emulate rawmode for keycode 240
[14440.604840] keyboard: can't emulate rawmode for keycode 240
[14440.604844] hp_wmi: Unknown key code - 0x21a5
[14440.832498] keyboard: can't emulate rawmode for keycode 240
[14440.832520] keyboard: can't emulate rawmode for keycode 240
[14440.832524] hp_wmi: Unknown key code - 0x21a5
[14444.577270] keyboard: can't emulate rawmode for keycode 240
[14444.577294] keyboard: can't emulate rawmode for keycode 240
[14444.577297] hp_wmi: Unknown key code - 0x21a5
[14444.968475] keyboard: can't emulate rawmode for keycode 240
[14444.968499] keyboard: can't emulate rawmode for keycode 240
[14444.968503] hp_wmi: Unknown key code - 0x21a5
[14445.127734] keyboard: can't emulate rawmode for keycode 240
[14445.127756] keyboard: can't emulate rawmode for keycode 240
[14445.127760] hp_wmi: Unknown key code - 0x21a5
[14485.377110] hp_wmi: Unknown key code - 0x21a5
[14486.752135] hp_wmi: Unknown key code - 0x21a5
[14487.352757] hp_wmi: Unknown key code - 0x21a5
[14487.791967] hp_wmi: Unknown key code - 0x21a5
[14488.217371] hp_wmi: Unknown key code - 0x21a5
[14610.840081] hp_wmi: Unknown key code - 0x21a5
```

If I try to assign <kbd>XF86Calculator</kbd> to this key, I get an error
```
sudo setkeycodes 21a5 140
setkeycodes: failed to set scancode 21a5 to keycode 140: ioctl KDSETKEYCODE: Invalid argument
```

pelrun, 2021-05-02:
Well, that's probably because the key mappings in that new commit are still commented out...

wl2776, 2021-05-02:
Have tried to uncomment line 176 (https://github.com/pelrun/hp-omen-linux-module/blob/master/module/hp-wmi.c#L176)
Have got touchpad disabled, and could not enable it back with Fn-F11

pelrun, 2021-05-02:
Ok, so somewhere 8 is getting added to the keycode set in the table, and that changes KEY_F23 to send 201 instead of 193. And `xmodmap -pke` shows keycode 201 is configured as XF86TouchpadOff. :1st_place_medal: 

I've figured out that we don't want `setkeycodes` at all, instead you use xmodmap to set what X11 keysym is triggered by a particular keycode (for instance KEY_CALC is mapped to `XF86Calculator`, which is actually what launches the thing.) I just need to get the driver correctly sending keycodes that are normally empty in the table.

pelrun, 2021-05-03:
Answers to my own questions: the first 8 (well, MIN_KEYCODE) X11 keycodes are reserved, so linux input subsystem keycodes have 8 added when they get into X. F23/F24 literally are set as touchpad off/on in the default mapping, it's not a bug.

I've mapped the hotkeys to F14-F17 now, which by default are XF86Launch5 to XF86Launch8. Just map them to applications using whatever desktop hotkey configuration tool your distro provides (for instance it's under Desktop->Keyboard in the ubuntu settings app.) No need to remap via xmodmap for most cases.

wl2776, 2021-05-03:
I've pulled your recent commits, and they didn't work.
<kbd>Omen</kbd> and <kbd>Calculator</kbd> keys now do nothing, and I cannot assign shortcuts in KDE settings.

`dmesg` doesn't show any error or warning messages about unknown events or keycodes.

Tried `sudo showkey -k ` and `sudo showkey -s`.
<kbd>Omen</kbd> doesn't produce any keycodes or scancodes.
<kbd>Calculator</kbd> produces scancodes 0xe0 0x21 and `keycode 140 press`.

wl2776, 2021-05-03:
I've added some debug messages in `hp_wmi_notify`. See head commit in my fork.
When I press <kbd>Omen</kbd> key, I see the following in dmesg:
```
hp_wmi: status = 0 value = 128 response = 32 000000002f4242ee
```
And no `Omen key` message.

pelrun, 2021-05-03:
It's just a very silly bug in the last commit - i used && in the release flag check instead of &.

wl2776, 2021-05-03:
https://github.com/wl2776/hp-omen-linux-module/commit/45b694b6c2a69a778267cdf964c94e6503c780f3 - <kbd>Omen</kbd> works good.

The rest is <kbd>Calculator</kbd>
When I press it, nothing appears in dmesg.

It produces following with `sudo showkey -k`
```
keycode 140 press
keycode 140 press
keycode 140 press
keycode 140 press
keycode 140 press
```

and following with `sudo showkey -s`
```
0xe0 0x21
0xe0 0x21
0xe0 0x21
```

However, I cannot assign it as a shortcut for calculator launch in KDE. 
I bring a shortcut assignment dialog, press that key, but nothing happens, KDE doesn't see any keypresses.
Although now it sees <kbd>Omen</kbd>

pelrun, 2021-05-03:
That calculator key is a normal media key, and is clearly being correctly mapped to XF86Calculator. That puts the issue squarely in KDE's court. A quick google brought up https://askubuntu.com/q/1199160 which might help (I have neither KDE or a kb with a calculator button which makes it hard for me to check...)

closed 2021-05-06 by pelrun

I-dont-need-name, 2021-08-23:
So the calculator button will not be fixed? It's not a KDE problem, this happens in every DE i tried.

pelrun, 2021-08-23:
That key doesn't get processed by this module *at all* - it uses the same mechanism as the ordinary keys. So there's nothing I can do here to fix it.

I-dont-need-name, 2021-08-24:
> That key doesn't get processed by this module _at all_ - it uses the same mechanism as the ordinary keys. So there's nothing I can do here to fix it.

Thank you for your reply. Indeed, it looks like the key is being processed along with all other keys on the keyboard. Could you please tell where I should report this bug then?

wl2776, 2021-08-24:
> Thank you for your reply. Indeed, it looks like the key is being processed along with all other keys on the keyboard. Could you please tell where I should report this bug then?

I've filed the bug report: https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/1929345

I-dont-need-name, 2021-10-06:
> I've filed the bug report: https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/1929345
to fix this you need to add this to /usr/lib/udev/hwdb.d/60-keyboard-hwdb
```
# HP Omen 15
evdev:atkbd:dmi:bvn*:bvr*:bd*:svnHP:pnOMENLaptop15*:pvr*
 KEYBOARD_KEY_a1=!calc
```
I submitted the fix to systemd already so it should be working ootb soon.

I-dont-need-name, 2021-10-10:
> Could you give me a link with description of these lines? `man udev` and `man systemd-hwdb` don't say much.
https://github.com/systemd/systemd/blob/main/hwdb.d/60-keyboard.hwdb

# An '!' as the first character of the key identifier string
# will add the scan code to the AT keyboard's list of scan codes
# where the driver will synthesize a release event and not expect
# it to be generated by the hardware.

xddxdd, 2022-02-22:
Just a note that the hwdb fix from @I-dont-need-name also applies to HP Omen 17 laptops, but with a different matching entry:
```text
evdev:atkbd:dmi:bvn*:bvr*:bd*:svnHP:pnOMENbyHPLaptop17*:pvr*
 KEYBOARD_KEY_a1=!calc
```

wl2776, 2022-02-22:
Then, maybe its possible to make it shorter and more universal

```
evdev:atkbd:dmi:bvn*:bvr*:bd*:svnHP:pnOMEN*:pvr*
 KEYBOARD_KEY_a1=!calc
```

xddxdd, 2022-02-22:
Yeah, I guess it's the common behavior for HP OMEN laptops. Just sent a PR to systemd, will see how that goes.

xrun1989, 2024-10-10:
I am trying to remap `OMEN` key on windows with SharpKeys. I don't remember where some user manage to do it! I know this is not related to Linux but I need to know what is the omen key code.

wl2776, 2024-10-10:
>  I need to know what key is `omen`.

Are you asking about [this](https://github.com/pelrun/hp-omen-linux-module/blob/9d1edb1d2b2f9bf6b5ac0ba188d7d421fc239d2b/src/hp-wmi.c#L71)?

pelrun, 2024-10-11:
Unfortunately `OMEN` isn't a real key and doesn't produce any keycodes when pressed, which means normal hotkey utilities won't work with it. On Linux we're intercepting the events it *does* generate and then faking a keypress, but that's not useful for figuring out how Windows is handling it.