pelrun/hp-omen-linux-module #13: I'm not seeing the expected 'rgb_zones' under hp-wmi.

leetwanker, 2021-06-05:
```
OMEN:~/Downloads/hp-omen-linux-module-master$ sudo make install
dkms install .

Creating symlink /var/lib/dkms/hp-omen-wmi/0.9/source ->
                 /usr/src/hp-omen-wmi-0.9

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j12 KERNELRELEASE=5.8.0-55-generic -C src/ KERNELDIR=/lib/modules/5.8.0-55-generic/build...
Signing module:
Generating a new Secure Boot signing key:
Can't load /var/lib/shim-signed/mok/.rnd into RNG
139764788532544:error:2406F079:random number generator:RAND_load_file:Cannot open file:../crypto/rand/randfile.c:98:Filename=/var/lib/shim-signed/mok/.rnd
Generating a RSA private key
.................................+++++
...................................................................................................................................................+++++
writing new private key to '/var/lib/shim-signed/mok/MOK.priv'
-----
 - /var/lib/dkms/hp-omen-wmi/0.9/5.8.0-55-generic/x86_64/module/hp-wmi.ko
Secure Boot not enabled on this system.
cleaning build area...

DKMS: build completed.

hp-wmi.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.8.0-55-generic/updates/dkms/

depmod...

DKMS: install completed.
```
Looks like everything installed fine? Any ideas? HP OMEN Model:  17-cb1055cl

Thanks for your work on this!

pelrun, 2021-06-05:
I think I recall it not automatically loading the module after it's installed, although it'll come up after a reboot.

Try `sudo modprobe hp-wmi` and see if that sorts it out.

leetwanker, 2021-06-05:
It didn't, but a reboot did! Working great. Is there a way to turn the backlight on/off in addition to changing the color? I'd like to have them turn on automatically when typing. 

Thanks

pelrun, 2021-06-05:
Not yet; that's the separate brightness control that's still on the todo list.

leetwanker, 2021-06-05:
Looking forward to it!

closed 2021-06-05 by leetwanker