pelrun/hp-omen-linux-module #14: Would it be a big difference if I test this software in a VM

EfraMP, 2021-08-11:
According to the description of the project, this is experimental and could crash my machine.

What if I just test it on a VM? Would it be a great difference? 
I just ask cause I don't have a lot of experience on VM's

markozajc, 2021-08-11:
I'm pretty sure virtual machines don't expose the hardware interfaces required to run this module. 
Either way, you can just live boot and remove the module if the machine fails to boot. Unless it fries it, which is probably not going to happen.

EfraMP, 2021-08-11:
Sound very reasonable and smart to me. Thanks!

pelrun, 2021-08-11:
It's mostly a boilerplate disclaimer as I've not done any real testing. But I haven't had any reports of problems.

EfraMP, 2021-08-11:
Yeah, I know it's just a disclaimer. Thanks!

closed 2022-01-07 by pelrun