pelrun/hp-omen-linux-module #16: Problems with other HP Omen models

Doumor, 2021-11-28:
In another issue I read that the app interface is different for different models. Could you implement it for my model?

OMEN Laptop 15-en0xxx(15-en0043ur)

mail: doumor@vk.com
I'll send you a dump and a copy of this message from this post.

alexkutsan, 2022-09-18:
Does not work for my model as well. How to know what is my model? 

```
alex@alex-OMEN:~$ sudo dmidecode | grep -A 9 "System Information"
System Information
	Manufacturer: HP
	Product Name: OMEN by HP Laptop 15-dh1xxx
	Version:  
	Serial Number: CND0200741
	UUID: f41296a3-2a90-ea11-8104-b05cdad971b0
	Wake-up Type: Power Switch
	SKU Number: 16F76EA#ABD
	Family: 103C_5335KV HP OMEN
```

ahoneybun, 2022-09-23:
@alexkutsan it's usually on the bottom of the system I believe. This is mine:

```
System Information
	Manufacturer: HP
	Product Name: OMEN by HP Laptop 15-dh0xxx
	Version:  
	Serial Number: CND9386XL7
	UUID: 9e7f106e-71d3-e911-8102-040e3c547666
	Wake-up Type: Power Switch
	SKU Number: 8DT27UA#ABA
	Family: 103C_5335KV HP OMEN
```

ahoneybun, 2022-09-26:
What's odd is that it worked in Fedora but not Pop!_OS.

Doumor, 2023-01-19:
`System Information
        Manufacturer: HP
        Product Name: OMEN Laptop 15-en0xxx
        Version:  
        Serial Number: 5CD052LKNK
        UUID: 30444335-3235-4b4c-4e4b-3024a98fd16e
        Wake-up Type: Power Switch
        SKU Number: 24D54EA#ACB
        Family: 103C_5335KV HP OMEN
`

Doumor, 2023-01-19:
```
System Information
        Manufacturer: HP
        Product Name: OMEN Laptop 15-en0xxx
        Version:  
        Serial Number: 5CD052LKNK
        UUID: 30444335-3235-4b4c-4e4b-3024a98fd16e
        Wake-up Type: Power Switch
        SKU Number: 24D54EA#ACB
        Family: 103C_5335KV HP OMEN

```