pelrun/hp-omen-linux-module #18: Enabling of macro keys P1-P6

alexx7676, 2021-12-22:
I have a notebook HP Omen 17-cb1055cl. It has 6 macro keys P1-P6.
Is there exists any possibility to enable these keys in linux?

xddxdd, 2022-04-03:
I've done some reverse engineering on the OMEN Command Center app, and made these keys work on my system (OMEN 17t-ck000).

My working code is in <https://github.com/xddxdd/hp-omen-linux-module>, but I don't plan to send a PR until I make it more usable for regular users.

Notes:

1. HP has several revisions of macro key hardware, so this may not work for you, but it shouldn't do any harm to your hardware anyway.

2. Right now my module maps P1-P6 to numpad 1-6, Fn+P1-P4 to numpad 7,8,9,0, Fn+P5 to numpad minus, and Fn+P6 to numpad plus.
   1. I can't map them to non-standard keys (like F13-F24, or other advanced media keys supported by Linux), because the hardware doesn't support them.
   2. My laptop doesn't have a numpad, so there's no interference for me. But if yours do, then the keys will conflict, and because of (i) I can't do anything about it.
   3. If you just want to use them for shortcuts, this should be enough for you.

3. It's possible to send a series of keystrokes on keypress (just like what OMEN Command Center does), but currently this involves directly editing the source code of the module.

alker694, 2024-09-26:
i configured your module so it works with 6.11 kernels but the problem here is p2,p4 and p6 does not work 
the other keys work 

can you help me 

just tell me what is the problem and i will try to fix it my self 

thank you

mssvieira, 2024-12-11:
Hey @xddxdd any updates on this matter?  I'm running into this exact problem on a Omen-17cm2003ns and I'd like to just use them as simple shortcuts. 

If not, any leads on how I could debunk this for personal investigation and usage.