pelrun/hp-omen-linux-module #24: Any progress on the fans?

wagenadl, 2022-06-07:
Module mostly works on OMEN by HP Laptop 17-ck0xxx:

* Very happy to set the colors to something less garish. 
* The Omen key is available to xkb

What doesn't work:

* The P1 to P6 keys are not available to xkb.

Any progress on the fan control? That would be my major request: my laptop keeps its fan on all the time at a low speed and I'd much prefer silence...

Thanks!

I-dont-need-name, 2022-08-01:
For me nbfc works for fan control.

alou-S, 2022-08-15:
@I-dont-need-name Omen 15? I have a Omen 16 and nbfc doesn't seem to work

I-dont-need-name, 2022-08-16:
@alou-S yes. I used this config file: https://github.com/hirschmann/nbfc/issues/1227 (Slightly adjusted the values to make it more cool)

alou-S, 2023-03-12:
For those who come here later I made this Python Script that uses EC to control fans : https://github.com/alou-S/omen-fan