pelrun/hp-omen-linux-module #25: Brightness Control

rolfoz, 2022-06-09:
Firstly thank you for your work.
I have an Omen 16 and this module lets me control the zones and colors.
What I would like to do is adjust the led brightness and turn the leds on and off from a script.

I was hoping that using evtest I could replicate the FN F4 command that turns the leds on and off, but the FN button does not register.

If you do have any ideas how to turn off the keyboard leds when the screen turns off and how to control the brightness please let me know.

alou-S, 2022-07-30:
Brightness can simply be controlled by making the RGB values darker. And for turning off you can set the RGB values to #000000

Now implementing the backlight turn off on screen off is something you can try researching yourself as this will depend on various factors such as DE and others things.

rolfoz, 2022-08-01:
Awesome, thank you. I realized I could adjust the brightness with the values but it never occurred to me to completely turn them off that way!

closed 2022-08-01 by rolfoz