pelrun/hp-omen-linux-module #27: Bug in rgb zones 00 and 01

reipo38, 2023-11-04:
I think I've found a bug where if you try to change both zones 00 and 01 at the same time (for example with a single line in the terminal, with two commands separated by | or in a script that changes all zones to different colours at the same time), zone00 will take both the values for 00 and 01 (if you try to set 00 to 00FF00 and 01 to 0000FF, 01 will be changed to 0000FF and 00 will be changed to 00FFFF).

When I tried it in terminal sometimes 00 would take the appropriate value and sometimes the mixed value but completely at random.

When I tried to run a script that updates every zone with values from an array with three values using a loop, it'll at first take the correct value, then a mix from the one it should and the one 01 takes, and finally it'll take "000000"
like this:
array[]={1,2,3}

first iteration:
00->1
01->2
02->3

second iteration:
00->3+1
01->1
02->2

third iteration:
00->" "
01->3
02->1

pelrun, 2023-11-05:
Yes, that's not exactly a surprise as the current implementation isn't threadsafe. The simplest fix is to do the updates sequentially rather than simultaneously - i.e. don't use `|`, just run the commands one after the other. It shouldn't really be any slower, but it will avoid any races.

reipo38, 2023-11-05:
That's the intuitive thing to do however not practical in a script. I tried a lot of different syntaxes including using a nested loop to execute each change in its own iteration with both executing a shell script inside the program or directly editing the files. The only thing that fixed this issue was just executing the changes via shell script in reverse order. Strangely editing the files causes this issue in both editing from 00 to 02 and from 02 to 00. Shell script caused the issue only from 02 to 00. Very weird issue but I managed to work around it somewhat.

pelrun, 2023-11-05:
> not practical in a script

How exactly are you trying to do it? There's absolutely nothing "impractical" about executing multiple commands in a loop in bash.

reipo38, 2023-11-05:
I'm not doing it in bash but in python. Technically in all of my attempts I executed the commands sequentially but the issue still persists. I even added a 1 second wait after each update to see if it justs executes the commands too fast but it didn't help. This seems very illogical since the commands not only get executed one after the other but there's also a rather large gap between their executions.

ranisalt, 2024-07-19:
@reipo38 you need to use unbuffered I/O https://docs.python.org/3/library/io.html#raw-i-o