pelrun/hp-omen-linux-module #28: Arch linux-headers-6.11.9-arch1-1 [FIX]

CesarrSantos, 2024-11-21:
```
static  void  __exit hp_wmi_bios_remove(struct platform_device *device)
{
  int i;
  cleanup_sysfs(device);

  for (i = 0; i < rfkill2_count; i++) {
    rfkill_unregister(rfkill2[i].rfkill);
    rfkill_destroy(rfkill2[i].rfkill);
  }

  if (wifi_rfkill) {
    rfkill_unregister(wifi_rfkill);
    rfkill_destroy(wifi_rfkill);
  }
  if (bluetooth_rfkill) {
    rfkill_unregister(bluetooth_rfkill);
    rfkill_destroy(bluetooth_rfkill);
  }
  if (wwan_rfkill) {
    rfkill_unregister(wwan_rfkill);
    rfkill_destroy(wwan_rfkill);
  }

  //return 0;
}
```

hockeymikey, 2024-12-08:
Open a pull request?

JonnyTech, 2024-12-28:
Thanks, this work for me on Debian with kernel 6.11.7-amd64.