pelrun/hp-omen-linux-module #31: Error making for kernel 6.12

JonnyTech, 2024-12-29:
Getting errors trying to make with kernel 6.12:

```
$ uname -r
6.12.6-amd64

$ sudo make
dkms install .
Sign command: /lib/modules/6.12.6-amd64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/hp-omen-wmi/0.9/source -> /usr/src/hp-omen-wmi-0.9

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.12.6-amd64 -C src/ KERNELDIR=/lib/modules/6.12.6-amd64/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.12.6-amd64 (x86_64)
Consult /var/lib/dkms/hp-omen-wmi/0.9/build/make.log for more information.
make: *** [Makefile:2: install] Error 10

$ cat /var/lib/dkms/hp-omen-wmi/0.9/build/make.log
DKMS make.log for hp-omen-wmi-0.9 for kernel 6.12.6-amd64 (x86_64)
Sun 29 Dec 12:38:44 GMT 2024
make[1]: Entering directory '/var/lib/dkms/hp-omen-wmi/0.9/build/src'
make -C /lib/modules/6.12.6-amd64/build M=$PWD
make[2]: Entering directory '/usr/src/linux-headers-6.12.6-amd64'
  CC [M]  /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.o
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c: In function ‘hp_wmi_notify’:
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:556:12: error: implicit declaration of function ‘wmi_get_event_data’; did you mean ‘acpi_get_event_status’? [-Werror=implicit-function-declaration]
  556 |   status = wmi_get_event_data(value, &response);
      |            ^~~~~~~~~~~~~~~~~~
      |            acpi_get_event_status
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c: In function ‘hp_wmi_input_setup’:
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:709:57: error: passing argument 2 of ‘wmi_install_notify_handler’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  709 |   status = wmi_install_notify_handler(HPWMI_EVENT_GUID, hp_wmi_notify, NULL);
      |                                                         ^~~~~~~~~~~~~
      |                                                         |
      |                                                         void (*)(u32,  void *) {aka void (*)(unsigned int,  void *)}
In file included from /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:27:
/usr/src/linux-headers-6.12.6-common/include/linux/acpi.h:404:60: note: expected ‘wmi_notify_handler’ {aka ‘void (*)(union acpi_object *, void *)’} but argument is of type ‘void (*)(u32,  void *)’ {aka ‘void (*)(unsigned int,  void *)’}
  404 |                                         wmi_notify_handler handler, void *data);
      |                                         ~~~~~~~~~~~~~~~~~~~^~~~~~~
In file included from /usr/src/linux-headers-6.12.6-common/include/linux/printk.h:6,
                 from /usr/src/linux-headers-6.12.6-common/include/linux/kernel.h:31,
                 from /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:19:
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c: At top level:
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:1249:22: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Werror=incompatible-pointer-types]
 1249 |   .remove = __exit_p(hp_wmi_bios_remove),
      |                      ^~~~~~~~~~~~~~~~~~
/usr/src/linux-headers-6.12.6-common/include/linux/init.h:395:21: note: in definition of macro ‘__exit_p’
  395 | #define __exit_p(x) x
      |                     ^
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:1249:22: note: (near initialization for ‘hp_wmi_driver.<anonymous>.remove’)
 1249 |   .remove = __exit_p(hp_wmi_bios_remove),
      |                      ^~~~~~~~~~~~~~~~~~
/usr/src/linux-headers-6.12.6-common/include/linux/init.h:395:21: note: in definition of macro ‘__exit_p’
  395 | #define __exit_p(x) x
      |                     ^
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-6.12.6-common/scripts/Makefile.build:234: /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.o] Error 1
make[3]: *** [/usr/src/linux-headers-6.12.6-common/Makefile:1962: /var/lib/dkms/hp-omen-wmi/0.9/build/src] Error 2
make[2]: *** [/usr/src/linux-headers-6.12.6-common/Makefile:236: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.12.6-amd64'
make[1]: *** [Makefile:5: default] Error 2
make[1]: Leaving directory '/var/lib/dkms/hp-omen-wmi/0.9/build/src'
```

This is the result from fix for kernel 6.11 (#30):

```
$ sudo make
dkms install .
Sign command: /lib/modules/6.12.6-amd64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/hp-omen-wmi/0.9/source -> /usr/src/hp-omen-wmi-0.9

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.12.6-amd64 -C src/ KERNELDIR=/lib/modules/6.12.6-amd64/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.12.6-amd64 (x86_64)
Consult /var/lib/dkms/hp-omen-wmi/0.9/build/make.log for more information.
make: *** [Makefile:2: install] Error 10

$ cat /var/lib/dkms/hp-omen-wmi/0.9/build/make.log
DKMS make.log for hp-omen-wmi-0.9 for kernel 6.12.6-amd64 (x86_64)
Sun 29 Dec 12:43:16 GMT 2024
make[1]: Entering directory '/var/lib/dkms/hp-omen-wmi/0.9/build/src'
make -C /lib/modules/6.12.6-amd64/build M=$PWD
make[2]: Entering directory '/usr/src/linux-headers-6.12.6-amd64'
  CC [M]  /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.o
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c: In function ‘hp_wmi_notify’:
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:556:12: error: implicit declaration of function ‘wmi_get_event_data’; did you mean ‘acpi_get_event_status’? [-Werror=implicit-function-declaration]
  556 |   status = wmi_get_event_data(value, &response);
      |            ^~~~~~~~~~~~~~~~~~
      |            acpi_get_event_status
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c: In function ‘hp_wmi_input_setup’:
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:709:57: error: passing argument 2 of ‘wmi_install_notify_handler’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  709 |   status = wmi_install_notify_handler(HPWMI_EVENT_GUID, hp_wmi_notify, NULL);
      |                                                         ^~~~~~~~~~~~~
      |                                                         |
      |                                                         void (*)(u32,  void *) {aka void (*)(unsigned int,  void *)}
In file included from /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:27:
/usr/src/linux-headers-6.12.6-common/include/linux/acpi.h:404:60: note: expected ‘wmi_notify_handler’ {aka ‘void (*)(union acpi_object *, void *)’} but argument is of type ‘void (*)(u32,  void *)’ {aka ‘void (*)(unsigned int,  void *)’}
  404 |                                         wmi_notify_handler handler, void *data);
      |                                         ~~~~~~~~~~~~~~~~~~~^~~~~~~
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-6.12.6-common/scripts/Makefile.build:234: /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.o] Error 1
make[3]: *** [/usr/src/linux-headers-6.12.6-common/Makefile:1962: /var/lib/dkms/hp-omen-wmi/0.9/build/src] Error 2
make[2]: *** [/usr/src/linux-headers-6.12.6-common/Makefile:236: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.12.6-amd64'
make[1]: *** [Makefile:5: default] Error 2
make[1]: Leaving directory '/var/lib/dkms/hp-omen-wmi/0.9/build/src'
```

Does anyone have any suggestions?

JonnyTech, 2024-12-29:
According to https://docs.kernel.org/6.11/driver-api/wmi.html#c.wmi_get_event_data

```
acpi_status wmi_get_event_data(u32 event, struct acpi_buffer *out)

    Get WMI data associated with an event (deprecated)
```

`wmi_get_event_data` was deprecated in 6.11 and removed in 6.12 - not sure what to use from https://docs.kernel.org/6.12/driver-api/wmi.html

ATUK-OoD, 2024-12-31:
I'm having the exact same issue and behaviour. Waiting for some fix.

prasoondev, 2025-01-21:
```
static void hp_wmi_notify(union acpi_object *obj, void *context)
{
  struct acpi_buffer response = { ACPI_ALLOCATE_BUFFER, NULL };
  u32 event_id, event_data;
  u32 *location;
  int key_code;
  acpi_status status;

  // Replace wmi_get_event_data with acpi_evaluate_object
  char method[16];
  snprintf(method, sizeof(method), "_WED"); // Replace "_WED" with the actual ACPI method if different
  status = acpi_evaluate_object(NULL, method, NULL, &response);

  if (status == AE_NOT_FOUND)
  {
    // We've been woken up without any event data
    // Some models do this when the Omen hotkey is pressed
    event_id = HPWMI_OMEN_KEY;
  }
  else if (status != AE_OK) {
    pr_info("bad event value 0x%x status 0x%x\n", 0, status);
    return;
  }
  else
  {
    obj = (union acpi_object *)response.pointer;

    if (!obj)
      return;
    if (obj->type != ACPI_TYPE_BUFFER) {
      pr_info("Unknown response received %d\n", obj->type);
      kfree(obj);
      return;
    }

    /*
     * Depending on ACPI version the concatenation of id and event data
     * inside _WED function will result in a 8 or 16 byte buffer.
     */
    location = (u32 *)obj->buffer.pointer;
    if (obj->buffer.length == 8) {
      event_id = *location;
      event_data = *(location + 1);
    } else if (obj->buffer.length == 16) {
      event_id = *location;
      event_data = *(location + 2);
    } else {
      pr_info("Unknown buffer length %d\n", obj->buffer.length);
      kfree(obj);
      return;
    }
    kfree(obj);
  }

  switch (event_id) {
    case HPWMI_DOCK_EVENT:
      if (test_bit(SW_DOCK, hp_wmi_input_dev->swbit))
        input_report_switch(hp_wmi_input_dev, SW_DOCK,
                            hp_wmi_hw_state(HPWMI_DOCK_MASK));
        if (test_bit(SW_TABLET_MODE, hp_wmi_input_dev->swbit))
          input_report_switch(hp_wmi_input_dev, SW_TABLET_MODE,
                              hp_wmi_hw_state(HPWMI_TABLET_MASK));
          input_sync(hp_wmi_input_dev);
        break;
    case HPWMI_PARK_HDD:
      break;
    case HPWMI_SMART_ADAPTER:
      break;
    case HPWMI_BEZEL_BUTTON:
    case HPWMI_OMEN_KEY:
      key_code = hp_wmi_read_int(HPWMI_HOTKEY_QUERY);
      // Some hotkeys generate both press and release events
      // Just drop the release events.
      if (key_code < 0 || (key_code & HPWMI_HOTKEY_RELEASE_FLAG))
        break;

    if (!sparse_keymap_report_event(hp_wmi_input_dev,
      key_code, 1, true))
      pr_info("Unknown key code - 0x%x\n", key_code);
    break;
    case HPWMI_WIRELESS:
      if (rfkill2_count) {
        hp_wmi_rfkill2_refresh();
        break;
      }

      if (wifi_rfkill)
        rfkill_set_states(wifi_rfkill,
                          hp_wmi_get_sw_state(HPWMI_WIFI),
                          hp_wmi_get_hw_state(HPWMI_WIFI));
        if (bluetooth_rfkill)
          rfkill_set_states(bluetooth_rfkill,
                            hp_wmi_get_sw_state(HPWMI_BLUETOOTH),
                            hp_wmi_get_hw_state(HPWMI_BLUETOOTH));
          if (wwan_rfkill)
            rfkill_set_states(wwan_rfkill,
                              hp_wmi_get_sw_state(HPWMI_WWAN),
                              hp_wmi_get_hw_state(HPWMI_WWAN));
            break;
    case HPWMI_CPU_BATTERY_THROTTLE:
      pr_info("Unimplemented CPU throttle because of 3 Cell battery event detected\n");
      break;
    case HPWMI_LOCK_SWITCH:
      break;
    case HPWMI_LID_SWITCH:
      break;
    case HPWMI_SCREEN_ROTATION:
      break;
    case HPWMI_COOLSENSE_SYSTEM_MOBILE:
      break;
    case HPWMI_COOLSENSE_SYSTEM_HOT:
      break;
    case HPWMI_PROXIMITY_SENSOR:
      break;
    case HPWMI_BACKLIT_KB_BRIGHTNESS:
      break;
    case HPWMI_PEAKSHIFT_PERIOD:
      break;
    case HPWMI_BATTERY_CHARGE_PERIOD:
      break;
    default:
      pr_info("Unknown event_id - %d - 0x%x\n", event_id, event_data);
      break;
  }
}
```

Replace the function with this works for 6.12.x

ATUK-OoD, 2025-01-21:
Thanks for the solution, but it still was not working, althrouth different error now was:

```
DKMS make.log for hp-omen-wmi/0.9 for kernel 6.12.9-arch1-1 (x86_64)
Tue Jan 21 04:55:30 PM CET 2025
make[1]: Entering directory '/var/lib/dkms/hp-omen-wmi/0.9/build/src'
make -C /lib/modules/6.12.9-arch1-1/build M=$PWD
make[2]: Entering directory '/usr/lib/modules/6.12.9-arch1-1/build'
  CC [M]  /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.o
In file included from ./include/linux/printk.h:6,
                 from ./include/linux/kernel.h:31,
                 from /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:19:
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:1315:24: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Wincompatible-pointer-types]
 1315 |     .remove = __exit_p(hp_wmi_bios_remove),
      |                        ^~~~~~~~~~~~~~~~~~
./include/linux/init.h:395:21: note: in definition of macro ‘__exit_p’
  395 | #define __exit_p(x) x
      |                     ^
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:1315:24: note: (near initialization for ‘hp_wmi_driver.<anonymous>.remove’)
 1315 |     .remove = __exit_p(hp_wmi_bios_remove),
      |                        ^~~~~~~~~~~~~~~~~~
./include/linux/init.h:395:21: note: in definition of macro ‘__exit_p’
  395 | #define __exit_p(x) x
      |                     ^
make[4]: *** [scripts/Makefile.build:229: /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.o] Error 1
make[3]: *** [/usr/lib/modules/6.12.9-arch1-1/build/Makefile:1937: /var/lib/dkms/hp-omen-wmi/0.9/build/src] Error 2
make[2]: *** [Makefile:224: __sub-make] Error 2
make[2]: Leaving directory '/usr/lib/modules/6.12.9-arch1-1/build'
make[1]: *** [Makefile:5: default] Error 2
make[1]: Leaving directory '/var/lib/dkms/hp-omen-wmi/0.9/build/src'
```

But after playing with it for a while, I changed the "hp_wmi_bios_remove" to void intead of int, and removed/commented the "return 0;" at the end. 
Now it seems to work fine.

I'm not suggesting that this is the "intended" fix for it. I kinda still have no idea what I'm doing, but it works, seems fine. If there is some critical issue with it. Please tell me.

At the end the "hp_wmi_bios_remove" function on 1176'th line looks like this:

```
static void __exit hp_wmi_bios_remove(struct platform_device *device)
{
  int i;
  cleanup_sysfs(device);

  for (i = 0; i < rfkill2_count; i++)
  {
    rfkill_unregister(rfkill2[i].rfkill);
    rfkill_destroy(rfkill2[i].rfkill);
  }

  if (wifi_rfkill)
  {
    rfkill_unregister(wifi_rfkill);
    rfkill_destroy(wifi_rfkill);
  }
  if (bluetooth_rfkill)
  {
    rfkill_unregister(bluetooth_rfkill);
    rfkill_destroy(bluetooth_rfkill);
  }
  if (wwan_rfkill)
  {
    rfkill_unregister(wwan_rfkill);
    rfkill_destroy(wwan_rfkill);
  }

  // return 0;
}
```

prasoondev, 2025-01-21:
This issue first came up for 6.11.x so I already had it fixed in my code so just mentioned the changes needed to go from 6.11.x to 6.12.x

JonnyTech, 2025-01-21:
I have successfully built the module without errors but the omen key does not respond, even after a reboot:

```
$ uname -r
6.12.8-amd64

$ sudo make 
dkms install .
/usr/sbin/dkms.mx install . -k 6.12.8-amd64 -a x86_64
Sign command: /lib/modules/6.12.8-amd64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/hp-omen-wmi/0.9/source -> /usr/src/hp-omen-wmi-0.9

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.12.8-amd64 -C src/ KERNELDIR=/lib/modules/6.12.8-amd64/build...
Signing module /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.ko
Cleaning build area...

hp-wmi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.12.8-amd64/updates/dkms/
depmod.....

$ cat /var/lib/dkms/hp-omen-wmi/0.9/6.12.8-amd64/x86_64/log/make.log 
DKMS make.log for hp-omen-wmi-0.9 for kernel 6.12.8-amd64 (x86_64)
Tue 21 Jan 17:29:26 GMT 2025
make[1]: Entering directory '/var/lib/dkms/hp-omen-wmi/0.9/build/src'
make -C /lib/modules/6.12.8-amd64/build M=$PWD
make[2]: Entering directory '/usr/src/linux-headers-6.12.8-amd64'
  CC [M]  /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.o
  MODPOST /var/lib/dkms/hp-omen-wmi/0.9/build/src/Module.symvers
WARNING: modpost: /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi: section mismatch in reference: hp_wmi_driver+0x8 (section: .data) -> hp_wmi_bios_remove (section: .exit.text)
  CC [M]  /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.mod.o
  CC [M]  /var/lib/dkms/hp-omen-wmi/0.9/build/src/.module-common.o
  LD [M]  /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.ko
  BTF [M] /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.ko
Skipping BTF generation for /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.ko due to unavailability of vmlinux
make[2]: Leaving directory '/usr/src/linux-headers-6.12.8-amd64'
make[1]: Leaving directory '/var/lib/dkms/hp-omen-wmi/0.9/build/src'
```

Is this anything to worry about?

```
WARNING: modpost: /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi: section mismatch in reference: hp_wmi_driver+0x8 (section: .data) -> hp_wmi_bios_remove (section: .exit.text)
```

prasoondev, 2025-01-21:
Trying to figure out the same had the issue in 6.11.x aswell have been using 6.10.x due to suspend issues switched over to 6.12.x today did it work on 6.11.x?

prasoondev, 2025-01-21:
Can you retry after removing ```__exit``` from the function header?

JonnyTech, 2025-01-21:
> Can you retry after removing `__exit` from the function header? See if the warning presists

Yes, this stops the warning.

But the omen key still does not respond.

When I press it, this is logged:

```
$ sudo dmesg |tail -n 5
[  205.209360] hp_wmi: bad event value 0x0 status 0x1001
[  205.399361] hp_wmi: bad event value 0x0 status 0x1001
[  205.574452] hp_wmi: bad event value 0x0 status 0x1001
[  205.761361] hp_wmi: bad event value 0x0 status 0x1001
[  205.933335] hp_wmi: bad event value 0x0 status 0x1001
```

prasoondev, 2025-01-22:
Pretty sure its due to some kernel issue can you try after removing the module

prasoondev, 2025-01-22:
I might have a fix. When the key is pressed it triggers an event you can check which event by installing ```acpid``` then running ```acpi_listen``` note down the event when the key is pressed then make a script to launch some application when the event is triggered like I did for steam.
Enable the service acpid and start it before doing ``` acpi_listen```
event:
```
prasoon@fedora:~$ acpi_listen
battery PNP0C0A:00 00000081 00000001
wmi PNP0C14:01 00000080 00000000
```
wmi was the key press
so I wanted steam to launch when key is pressed so I made a script with its content being
```
#!/bin/bash
#Launch Steam
sudo -u prasoon steam &
```
now make this script executeable (sudo -u username because steam was made to be launched as a non-root)
not we make another script which looks for a event and executes our first script 
```
sudo nano /etc/acpi/events/omen_key
```
this creates a file in which we will handle the key press 
```
event=wmi PNP0C14:01 00000080 00000000
action=/path/to/launch_steam.sh

```
Now restart the service and try if it works.

For debug:
```
sudo systemctl stop acpid
acpid -d
```
Note: the event might be different for your machine.

JonnyTech, 2025-01-22:
Thanks @prasoondev but why would the behaviour change from previous versions?

Previously the module would allow the omen key to be recognised by the system as a standard keypress so that I could assign a shortcut in the operating system's default keyboard shortcut handler. It was very simple and straightforward and required no additional scripting.

prasoondev, 2025-01-22:
I did not test without the module on the newer kernels and the older kernels did not had the issue for the module. I suggest trying the key after removing the module if possible because i suspect the kernel for the issue and not the module.

JonnyTech, 2025-01-22:
When removing the module then the omen key is not detected at all, as per older kernels too.

prasoondev, 2025-01-22:
The key was fine for me till 6.10.x never seemed to work after 6.10.x

JonnyTech, 2025-01-22:
I fixed the omen key in my branch at https://github.com/JonnyTech/hp-omen-linux-module/tree/JonnyTech-kernel612 and submitted PR https://github.com/pelrun/hp-omen-linux-module/pull/32

JonnyTech, 2025-02-11:
@prasoondev thanks, I am aware of that repo, but unfortunately it makes my system unstable.

Lupus47, 2025-06-11:
I'm also getting a make issue on Manjaro with Kernel 6.12.28 with make being unsuccessful. Every mentioned fix in this thread leads to the same issue:
``` 
$sudo make          
dkms install .
Deprecated feature: CLEAN (/usr/src/hp-omen-wmi-0.9/dkms.conf)
Creating symlink /var/lib/dkms/hp-omen-wmi/0.9/source -> /usr/src/hp-omen-wmi-0.9

Sign command: /usr/lib/modules/6.12.28-1-MANJARO/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module(s)...(bad exit status: 2)
Failed command:
make -j20 KERNELRELEASE=6.12.28-1-MANJARO -C src/ KERNELDIR=/lib/modules/6.12.28-1-MANJARO/build

Error! Bad return status for module build on kernel: 6.12.28-1-MANJARO (x86_64)
Consult /var/lib/dkms/hp-omen-wmi/0.9/build/make.log for more information.
make: *** [Makefile:2: install] Error 10
```
With a log report of:
```
DKMS (dkms-3.2.1) make.log for hp-omen-wmi/0.9 for kernel 6.12.28-1-MANJARO (x86_64)
Tue Jun 10 09:27:38 PM CDT 2025

Building module(s)
# command: make -j20 KERNELRELEASE=6.12.28-1-MANJARO -C src/ KERNELDIR=/lib/modules/6.12.28-1-MANJARO/build
make[1]: Entering directory '/var/lib/dkms/hp-omen-wmi/0.9/build/src'
make -C /lib/modules/6.12.28-1-MANJARO/build M=$PWD
make[2]: Entering directory '/usr/lib/modules/6.12.28-1-MANJARO/build'
  CC [M]  /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.o
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c: In function ‘hp_wmi_notify’:
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:556:12: error: implicit declaration of function ‘wmi_get_event_data’; did you mean ‘acpi_get_event_status’? [-Wimplicit-function-declaration]
  556 |   status = wmi_get_event_data(value, &response);
      |            ^~~~~~~~~~~~~~~~~~
      |            acpi_get_event_status
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c: In function ‘hp_wmi_input_setup’:
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:709:57: error: passing argument 2 of ‘wmi_install_notify_handler’ from incompatible pointer type [-Wincompatible-pointer-types]
  709 |   status = wmi_install_notify_handler(HPWMI_EVENT_GUID, hp_wmi_notify, NULL);
      |                                                         ^~~~~~~~~~~~~
      |                                                         |
      |                                                         void (*)(u32,  void *) {aka void (*)(unsigned int,  void *)}
In file included from /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:27:
./include/linux/acpi.h:404:60: note: expected ‘wmi_notify_handler’ {aka ‘void (*)(union acpi_object *, void *)’} but argument is of type ‘void (*)(u32,  void *)’ {aka ‘void (*)(unsigned int,  void *)’}
  404 |                                         wmi_notify_handler handler, void *data);
      |                                         ~~~~~~~~~~~~~~~~~~~^~~~~~~
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:547:13: note: ‘hp_wmi_notify’ declared here
  547 | static void hp_wmi_notify(u32 value, void *context)
      |             ^~~~~~~~~~~~~
./include/linux/acpi.h:391:16: note: ‘wmi_notify_handler’ declared here
  391 | typedef void (*wmi_notify_handler) (union acpi_object *data, void *context);
      |                ^~~~~~~~~~~~~~~~~~
In file included from ./include/linux/printk.h:6,
                 from ./include/linux/kernel.h:31,
                 from /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:19:
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c: At top level:
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:1249:22: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Wincompatible-pointer-types]
 1249 |   .remove = __exit_p(hp_wmi_bios_remove),
      |                      ^~~~~~~~~~~~~~~~~~
./include/linux/init.h:395:21: note: in definition of macro ‘__exit_p’
  395 | #define __exit_p(x) x
      |                     ^
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:1249:22: note: (near initialization for ‘hp_wmi_driver.<anonymous>.remove’)
 1249 |   .remove = __exit_p(hp_wmi_bios_remove),
      |                      ^~~~~~~~~~~~~~~~~~
./include/linux/init.h:395:21: note: in definition of macro ‘__exit_p’
  395 | #define __exit_p(x) x
      |                     ^
/var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.c:1176:19: note: ‘hp_wmi_bios_remove’ declared here
 1176 | static int __exit hp_wmi_bios_remove(struct platform_device *device)
      |                   ^~~~~~~~~~~~~~~~~~
make[4]: *** [scripts/Makefile.build:229: /var/lib/dkms/hp-omen-wmi/0.9/build/src/hp-wmi.o] Error 1
make[3]: *** [/usr/lib/modules/6.12.28-1-MANJARO/build/Makefile:1949: /var/lib/dkms/hp-omen-wmi/0.9/build/src] Error 2
make[2]: *** [Makefile:224: __sub-make] Error 2
make[2]: Leaving directory '/usr/lib/modules/6.12.28-1-MANJARO/build'
make[1]: *** [Makefile:5: default] Error 2
make[1]: Leaving directory '/var/lib/dkms/hp-omen-wmi/0.9/build/src'

# exit code: 2
# elapsed time: 00:00:01
----------------------------------------------------------------
```
Any possible solutions?

prasoondev, 2025-06-12:
@Lupus47 try this
https://github.com/ranisalt/hp-omen-linux-module/tree/rebase-6.12

prasoondev, 2025-07-18:
@Lupus47 can you please mention which branch and what error message you got

Lupus47, 2025-08-27:
I have tried all suggested forks and have gotten the same error for each. Should I switch kernels to 6.15 if 6.12 is the problem?

Lupus47, 2025-09-28:
Is there a way to clear all the code from this repo easily? It's been a hot minute and I want to get this working